Schildkroet/Candle2 #1: Can I use Hightmap?

zubon2003, 2019-08-31:
Hello Schildkroet.
Today.I tested Candle2.
And I can't enter  Hightmap edit mode.
"Use heightmap" checkbox always grayed out.So I can't check it.

Schildkroet, 2019-09-02:
Since i don't use it, i disabled it.

In frmmain.cpp uncomment line 530 and comment out line 531. Then it should work.

`ui->widgetHeightMap->setEnabled(!m_processingFile && m_programModel.rowCount() > 1);`
`//ui->widgetHeightMap->setEnabled(false);`

zubon2003, 2019-09-02:
Thanks.I did it.

closed 2019-09-03 by zubon2003

czapeczek, 2019-10-17:
I am running through QT tutorials, but there's no clear way for me to compile this correctly. I mean - I do compile and the app runs, but in the project directory there's only candle2.exe file created without dll's. 

How to compile the application with modified code so that it is runnable?

Schildkroet, 2019-10-17:
You would have to copy the DLLs manually from the install dir.
But the easiest way is to download the current version and just replace the .exe with your own one.

czapeczek, 2019-10-17:
I tried that. I get an error:
The procedure entry point 
_ZN10QByteArray14trimmed_helperERS_ could not be located in the dynamic link library

Sorry but this doesn't tell me anything

Schildkroet, 2019-10-17:
Did you compile in release mode with Qt 5.4.2?

czapeczek, 2019-10-17:
Qt Creator 4.10.1
Based on Qt 5.13.1 (MSVC 2017, 32 bit)

so probably not. but I used online installator for open source projects, I don't recall I could choose a version there.

czapeczek, 2019-10-17:
I found some directory with 5.4.2 version. Will give it a shot