Schildkroet/Candle2 #3: Pressing STOP during milling causes damage to material and drill

czapeczek, 2019-10-18:
I am confused. I am sure it didn't work this way before but I am not able to track this to source. 
When I am pressing STOP button during milling process, the machine goes to 0,0 with it's current Z position which is the milling position. The spindle stop and it craches drill cause it's going trough the unmilled material. 

I was looking on many things like safe position gcode or anything else. To me pressing stop should just stop everything without moving back to x0y0

Schildkroet, 2019-10-18:
That's a "feature" of Candle.
I also noticed the behaviour sometimes, but didn't have time to analyze it yet.

czapeczek, 2019-10-19:
The problem is caused by function:

`void frmMain::restoreOffsets()
{
    // Still have pre-reset working position
    sendCommand(QString("G21G53G90X%1Y%2Z%3").arg(toMetric(ui->txtMPosX->text().toDouble()))
                                       .arg(toMetric(ui->txtMPosY->text().toDouble()))
                                       .arg(toMetric(ui->txtMPosZ->text().toDouble())), -1, m_settings->showUICommands());

    sendCommand(QString("G21G92X%1Y%2Z%3").arg(toMetric(ui->txtWPosX->value()))
                                       .arg(toMetric(ui->txtWPosY->value()))
                                       .arg(toMetric(ui->txtWPosZ->value())), -1, m_settings->showUICommands());
}`
in frmmain.cpp
It's ok to just comment those commands but it would be good to understand the intention why there are there. This function is used only twice in code, but really only once depending on the protocol used. 

For now I can just comment that and make a pull request, just let me know

Schildkroet, 2019-10-22:
I decided to comment the function call out. Restoring the offset, after you press STOP seems unnecessary and is dangerously.
Thanks for you help

closed 2019-10-22 by Schildkroet