Schildkroet/Candle2 #5: Height Map Disabled

ntelectronics, 2020-04-14:
Hi,
I am running Candle2 with a 3018 machine to make PCB's.

I am unable to use the height map as it is disabled.
I have loaded a PCB Gcode file and that is OK onscreen.
If I run the code the router works perfectly.
Probe also works fine.

Can you advise me how to use/enable height maps.

Thanks
Allen Brroks
Snr Consultant
NTElectronics UK

Schildkroet, 2020-04-14:
Hello,
it is currently disabled, but i will enable it in the next version.
See https://github.com/Schildkroet/Candle2/issues/1 for a solution

ntelectronics, 2020-04-14:
Hi,
thanks for your quick reply.
Do you have a timescale on the next version?

Allen

Schildkroet, 2020-04-14:
No, there is no timescale. It comes when it comes. But i plan a new release the next few weeks.

closed 2020-04-14 by Schildkroet