Schildkroet/Candle2 #6: program dont start up at all !! dead end ?? old dropped project ?

OZ2CPU-thomas, 2020-05-15:
this dont run at all..
windows version, i try it on win 7 home..
no startup.

Schildkroet, 2020-05-15:
Hello,
which version did you run?
I tested it on Win 7 & 10 64bit. On both it works. And it seems to run well on other PCs. So it seems to be specific to your PC.
Could you be more clear, what exactly you are doing?

OZ2CPU-thomas, 2020-05-15:
downloaded the latest version zip,
unzipped, go into folder, then try to run the exe file.. 
nothing at all, the pc win 7 home pc, seems to run anything else..

Schildkroet, 2020-05-15:
Are you sure, some anti virus etc. isn't interfering with execution? Does your system meet the requirements for running?

OZ2CPU-thomas, 2020-05-15:
Does your system meet the requirements for running?
good point , it is only a dual core pentium 2.1GHz with 4GB ram, running windows 7 home,
all other programs run fine on it, I prefer to use an old outdated pc for the milling machine,
so it dont matter if i get saw dust in it.

OZ2CPU-thomas, 2020-05-17:
still no luck, no one know why this program dont start ?

Schildkroet, 2020-05-17:
Well, since there's no error at all, i can't do that much. I can just assume there's a problem with your hardware or software. You could try downloading Qt and compile it yourself. This way you might get an error output, if it's still failing.

quietbiker, 2020-05-18:
The Qt library needs OpenGL2.  I was looking to run this on an old NetTop PC,  but that didn't support OpenGL2
It is an issue on the main Candle project, so assume it's the same on Candel 2:
https://github.com/Denvi/Candle/issues/108

Check the spec of your PC and perhaps provide more information.
I've since purchased a fairly cheap NetTop (some AcePC thing of Amazon) which runs it fine.

OZ2CPU-thomas, 2020-05-18:
installed new gfx drivers, now i could start candle 2.0 and connect to my machine,
even JOG works, visualisation works, default is black on black, not so smart..
how ever it refuse to start again, if I close the program !
then delete the settings file, then it starts, odd bug, but this is the same bug i see with all other versions of candle, need to delete the settings file to make it start, 
and none of the versions say anything about what is wrong ??
it made the settings file it self, so why cant it read its own setting file..
a good program, must always tell what is wrong, why it can not start or work.

Schildkroet, 2020-05-18:
Do you get an error after restart?

Try deleting these lines in settings and see if it works:
`antialiasing=true
msaa=true
vsync=false
zBuffer=false`

OZ2CPU-thomas, 2020-05-18:
nope, no errors, no info, just nothing happens, except a very short round circle blink in the cursor
I try to change my 3D settings to 16 bits / 32 bits, all sorts of random experiments,
no luck to start candle 2.0 at the moment. even if I deleted my settings file.
right now I am back to candle 1.17, seems to start only with settings file deleted,
but then it connect to my machine, and its visulalizer works great too,
I tried other programs like UGS, it works perfectly, its visualizer works too,
but it is really a lame program with a lot of the most important features compleetly missing
so i really need to get candle working. I got another new and sexy pc, where all versions of candle works, it is win10, but I really hate to get sawdust and metal parts flying into this machine,
it cost more than my milling machine.. sad day,,

closed 2020-06-01 by Schildkroet