Schildkroet/Candle2 #9: Compare Candle 1 and Candle 2 in Arduino Nano and issues

gabrail-110, 2020-09-14:
Plug 2: CNC motion on the straight line is excellent and on the arc is weak and on the curved line has high vibration and slow motion.
And (device settings) does not work, but has advanced features such as Handweel and Ethernet
Candle 1: All movements of line, arc and curve have excellent performance and Candle 1 works well but lacks advanced features like Candle 2

gabrail-110, 2020-09-15:
arc is weak and on the curved line has high vibration and slow motion.

Schildkroet, 2020-09-27:
Sent time of gcode is slower in Candle2. If your gcode is made out of many very very short moves this might be seen on grbl. I will hopefully improve it in next version.
What device settings are you talking about?

gabrail-110, 2020-10-02:
What device settings are you talking about?
[image]

gabrail-110, 2020-10-02:
test on candle 2 ( line-arc-curve)
[image]
time elapsed
[image]

Schildkroet, 2020-10-02:
OK, the  machine settings are not working, because i didn't implement them yet. That's a feature for the future.

Schildkroet, 2020-10-02:
I think i see the problem. Can you attach the GCode file so i can test?

gabrail-110, 2020-10-02:
test on candle 1 ( line-arc-curve)  
GRBL settings are the same as candle 2
[image]

gabrail-110, 2020-10-02:
gcode file
G17
G0Z20.000
G0X0.000Y0.000S18000M3
G0X15.356Y-26.149Z5.000
G0Z0.000
G1Z-0.100F10000.0
G1Y-211.841
X31.356
X26.180Y-194.920
X50.787Y-177.002
X30.894Y-168.774
X41.246Y-134.931
X15.356Y-118.995
X47.391Y-95.204
X19.135Y-75.664
X31.356Y-58.368
Y-39.098
X15.356Y-26.149
G0Z5.000
G0X70.949Y-31.056
G0Z0.000
G1Z-0.100F10000.0
G2X67.188Y-79.961I-2.772J-24.384
G0Z5.000
G0Z0.000
G1Z-0.100F10000.0
G3X96.865Y-114.237I30.145J-3.885
G0Z5.000
G0Z0.000
G1Z-0.100F10000.0
G2X77.219Y-163.142I-18.318J-21.040
G0Z5.000
G0Z0.000
G1Z-0.100F10000.0
G3X108.987Y-194.073I15.884J-15.466
G0Z5.000
G0X87.042Y-152.689
G0Z0.000
G1Z-0.100F10000.0
G3X101.042Y-138.689I0.000J14.000
G3X87.042Y-124.689I-14.000J0.000
G3X73.042Y-138.689I0.000J-14.000
G3X87.042Y-152.689I14.000J0.000
G0Z5.000
G0X96.029Y-105.755
G0Z0.000
G1Z-0.100F10000.0
G3X108.029Y-93.755I0.000J12.000
G3X96.029Y-81.755I-12.000J0.000
G3X84.029Y-93.755I0.000J-12.000
G3X96.029Y-105.755I12.000J0.000
G0Z5.000
G0X74.711Y-72.971
G0Z0.000
G1Z-0.100F10000.0
G3X90.711Y-56.971I0.000J16.000
G3X74.711Y-40.971I-16.000J0.000
G3X58.711Y-56.971I0.000J-16.000
G3X74.711Y-72.971I16.000J0.000
G0Z5.000
G0X123.072Y-25.095
G0Z0.000
G1Z-0.100F10000.0
G1X124.490Y-25.478
X125.860Y-25.878
X127.181Y-26.296
X128.455Y-26.731
X129.680Y-27.184
X130.858Y-27.654
[... 1670 lines cut ...]
X162.866Y-10.675
X162.521Y-10.598
X162.154Y-10.553
X161.765Y-10.537
X161.394Y-10.553
X161.036Y-10.600
X160.691Y-10.679
X160.359Y-10.789
X160.040Y-10.930
X159.734Y-11.102
X159.442Y-11.306
X159.162Y-11.542
X158.902Y-11.803
X158.668Y-12.085
X158.459Y-12.388
X158.276Y-12.712
X158.119Y-13.056
X157.988Y-13.421
X157.883Y-13.807
X157.803Y-14.214
G0Z5.000
G0Z20.000
G0X0.000Y0.000
M2
%

gabrail-110, 2020-10-02:
file text

[test line arc curve.txt](https://github.com/Schildkroet/Candle2/files/5317468/test.line.arc.curve.txt)

Schildkroet, 2020-10-08:
I commited some improvements for speed you can test when compiling yourself

closed 2020-12-02 by Schildkroet