Schildkroet/Candle2 #12: The program does not start

ForestDog, 2021-03-07:
Win7 x64
"The resulting problem led to the termination of the program. Close this program.":
[image]
Windows information:
[image]
Other versions of Candlle (0.8.0 - 1.7.12) run without errors.
Thank.

UdoSchake, 2021-03-11:
Normal W7 updates. No additional libs.

Schildkroet, 2021-03-12:
Hello,
the last release was built with a newer QT version (5.15.2). An update corrupted the old one and i had to reinstall it.
So there might be an outdated package on your PC like .NET or MSVC++ Redistributable.
You could try removing the styles-folder

ForestDog, 2021-03-13:
Hello,
Unfortunately, this version of the program did not start on my second PC either. With the same error in the Windows log (the only difference is that the program did not start without displaying a window with an error message). Deleting the styles folder and the settings.ini file didn't change anything. Installed packages:

Microsoft .NET Framework 4.8.03761
Microsoft .NET Framework 4.8.03761
Update for Microsoft .NET Framework 4.8 (KB4503575) 1
Update for Microsoft .NET Framework 4.8 (KB4530746) 1
Microsoft Visual Basic / C ++ Runtime (x86) 1.1.0
Microsoft Visual C ++ 2005 Redistributable x64 8.0.61000
Microsoft Visual C ++ 2005 Redistributable x86 8.0.61001
Microsoft Visual C ++ 2008 Redistributable x64 9.0.30729.6161
Microsoft Visual C ++ 2008 Redistributable x86 9.0.30729.6161
Microsoft Visual C ++ 2010 Redistributable x64 10.0.30319
Microsoft Visual C ++ 2010 Redistributable x64 10.0.40219
Microsoft Visual C ++ 2010 Redistributable x86 10.0.30319
Microsoft Visual C ++ 2010 Redistributable x86 10.0.40219
Microsoft Visual C ++ 2012 Additional Runtime x64 11.0.61135
Microsoft Visual C ++ 2012 Minimum Runtime x64 11.0.61135
Microsoft Visual C ++ 2012 Additional Runtime x86 11.0.61135
Microsoft Visual C ++ 2012 Minimum Runtime x86 11.0.61135
Microsoft Visual C ++ 2013 Additional Runtime x64 12.0.40664
Microsoft Visual C ++ 2013 Minimum Runtime x64 12.0.40664
Microsoft Visual C ++ 2013 Additional Runtime x86 12.0.40664
Microsoft Visual C ++ 2013 Minimum Runtime x86 12.0.40664
Microsoft Visual C ++ 2015-2019 Redistributable x64 14.28.29325.2
Microsoft Visual C ++ 2019 Additional Runtime x64 14.29.29917
Microsoft Visual C ++ 2019 Minimum Runtime x64 14.29.29917
Microsoft Visual C ++ 2019 Additional Runtime x86 14.29.29917

I don't know if this information will be useful in the archive of the result of an attempt to run the program through NirSoft's SimpleProgramDebugger:

[report.zip](https://github.com/Schildkroet/Candle2/files/6134325/report.zip)
Thank!

Schildkroet, 2021-03-25:
This is compiled with an older version
https://gofile.io/d/uCGXdX

ForestDog, 2021-03-25:
Thank! This version starts without errors!

closed 2021-03-28 by Schildkroet