Schildkroet/Candle2 #14: Candle 2.2 has stopped working

gabrail-110, 2021-05-14:
[image]

Schildkroet, 2021-05-16:
Did you take a look at this?
https://github.com/Schildkroet/Candle2/issues/12

zab-ig, 2021-05-26:
I have a similar problem: the program does not start on any of the five computers.
Files are not available at link # 12 in the March 25th post.

Schildkroet, 2021-06-03:
New DL link
https://gofile.io/d/37XuPv

zab-ig, 2021-06-04:
Thank you. This version works.

closed 2021-06-04 by Schildkroet

wallybok, 2022-05-09:
> Thank you. This version works.

Can you pass it to me if you still have it?

Schildkroet, 2022-05-09:
Can you pls check, if this works?
https://gofile.io/d/0VEJMq

wallybok, 2022-05-09:
> Can you pls check, if this works? https://gofile.io/d/0VEJMq

The link works! Thank you so much!

zab-ig, 2022-05-11:
Hello. Checked on three different computers. The program started only on one (windows 10). On the other two it does not start (windows 10, windows 8.1). Apparently there are not enough files or programs.

> 
> 
>Can you pls check, if this works?
>https://gofile.io/d/0VEJMq