Schildkroet/Candle2 #16: please could you add field to add port?

pitersmit, 2021-08-03:
please could you add field to add port?, by the way how to edit and compile to change the port and generate a new exe, thank you

Schildkroet, 2021-08-07:
Hello,
what do you mean with "field port"?
This is a qt project. You can build it like every other qt project.

pitersmit, 2021-08-12:
field to place the ip port, to place the ip port within the software

pitersmit, 2021-08-12:
I think could be useful to place the ip port within the software manually, because no every body use the same ip port, thank you

ejhoness, 2024-03-06:
yes, tanks. and my not connect in port 30501,
[image]

commit 66a55696 (2024-03-07): Added option to set ip port. fixes #16
closed 2024-03-07 by Schildkroet