Schildkroet/Candle2 #18: not issue, just a question.

cheechNC, 2021-09-18:
Is it possible to use a uno board (grbl1.1) and a w5500 ethernet shield to run Candle2 via Ethernet?

Schildkroet, 2021-09-22:
No.
1. Uno has no memory left
2. No Pins for Shield left
3. Different voltage levels

closed 2021-09-22 by Schildkroet

cheechNC, 2021-09-22:
thanks. 

I thought you may be able to use this on top of a UNO <https://www.dfrobot.com/product-2370.html> since it has  a "pass through" pin bus..  I have not looked into it in detail;

Anyways, I have a cool IO isolation shield from south africa and I would love to be able to use that also with the nucleo board or another more powerfull arduino with Ethernet. https://www.yrcnc.com/product/arduino-cnc-isolation-shield/ 

This would be almost complete robust system if we could incorporate all. As then we would have robustness form the ethernet comms and shielding against emf/noise through the I/Os.

What do you think? :)

Cheers,
Frank

btw, Is there any connection diagram of your setup with the nucleo board?

Schildkroet, 2021-09-22:
You can find the pinout in the doc folder

I designed my own board with all features like eeprom, ethernet, 5v stepper output, 12v input ...
[image]

cheechNC, 2021-09-22:
Thanks Schildkroet impressive! :) Are you planning to sell these?
I like how the grbl trajectory planner behaves. It has quite a nice acceleration and deceleration behaviour and not bad path following. Although there is somewhat severe limitation to speed. My application only performs 3d Linear interpolation moves.. I mill large objects 2meters x 500mm x 150mm.  I could never get grbl to move faster then 250mm/s during 3d G1 moves. But still acceptable for a lower end machine. Still I need to solve the physical streaming interface.. Usb is asking for trouble. Ha
I wish I could design such a board.. very cool.Cheers.. thanks for sharing
Chico