Schildkroet/Candle2 #20: Does Candle 2.0 support grbl mega 5X?

jdeline, 2021-11-11:
I can connect to grbl mega 5x running on the Mega 2560 board, but can't seem to do anything once connected.

hankecnc, 2021-11-12:
> 我可以连接到在 Mega 2560 板上运行的 grbl mega 5x，但一旦连接似乎无法执行任何操作。

I am not the author, but I may help you, the possible situation is that the candle needs grbl message to be reported as grbl1.1f

Schildkroet, 2021-11-12:
Hello,
if grbl mega 5x uses the same protocol as normal grbl, it should work fine. But it should report back 1.1 as version. Just make sure you select corect com port, protocol and baudrate.

jdeline, 2021-11-12:
Using UGS (which works with my board), $I reports back [VER:1.2d.20210930:]

hankecnc, 2021-11-13:
> 1.2d
If you don’t know where the source code version report is, you can try to use a text search tool to search the source code folder, search for "1.2d", change it to "1.1f", recompile and upload.

Schildkroet, 2021-11-13:
You can also try to change the check in Candle2 and see if it works
https://github.com/Schildkroet/Candle2/blob/master/src/frmmain.cpp#L1725

closed 2021-11-13 by Schildkroet

fra589, 2021-11-14:
Hi all,  
  
It work when using the original check algorithm from https://github.com/Denvi/Candle frmmain.cpp:
```
bool frmMain::dataIsReset(QString data) {
    return QRegExp("^GRBL|GCARVIN\\s\\d\\.\\d.").indexIn(data.toUpper()) != -1;
}
```
_(I just tested it few minutes ago)_
This match any version of Grbl. I don't understand why you restrict Candle at the 1.1 version...
  
In Grbl-Mega-5X, the version string (1.2d) is at line 26 of the [grbl/grbl.h](https://github.com/fra589/grbl-Mega-5X/blob/edge/grbl/grbl.h#L26) file
I do not advise you to change this version string because I have made many changes to the original Grbl-Mega, this version chain allows you to handle the use of new functionalities by the different GCode GUI tools.
  
@++;  
Gauthier - Grbl-Mega-5X maintener

Schildkroet, 2021-11-15:
> This match any version of Grbl. I don't understand why you restrict Candle at the 1.1 version...

A change in version number might indicate a new protocol, thats the reason for the check

Is there a change in protocol for your 1.2d?

fra589, 2021-11-15:
Hi @Schildkroet, 

The protocol of grbl-Mega-5X is the same than the original Grbl.

But... :smiley: 

As Grbl-Mega-5X have capability of 3 to 6 axis, capacity to rename axis (change letter of axis) and capacity to clone (use 2 or more axis for the same letter), there may be some output differences:
- Depending of axis number, status reports can give from 3 to 6 axis position values,
- Axis parameters can be from 1x0 to 1x5,
- The $I output report [VER:], [AXS:] and [OPT:]. the AXS sentence report the axis number and axis names. For example, [AXS:5:XYZAC] mean there is 5 axis with names respectively X, Y, Z, A and C. The reported values of axis will be in this order. The [OPT:] output have been extended with some new values.
- The axis names can be any combination of X, Y, Z, A, B, C, U, V, W, D, E & H. XYZUVW are for linear axis and ABCDEH for rotationals.

The latest version is able to control up to 4 auxiliary digital output with GCode commands M62, M63, M64, M65 (see http://linuxcnc.org/docs/html/gcode/m-code.html#mcode:m62-m65).
So, the status report initiated by '?' is reporting digital output status in the accessory state (A:_SCFMD_) output when ther is one or more digital output active(s).

For more details, see the Grbl-Mega-5X interface Wiki : https://github.com/fra589/grbl-Mega-5X/wiki/grbl-Mega-5X-v1.2-interface.

@++;
Gauthier.