Schildkroet/Candle2 #22: For the drilling cycle G code, the subsequent omitted code, the opengl window should be drawn correctly

hankecnc, 2021-11-12:
For the drilling cycle G code, the subsequent omitted code, the opengl window should be drawn correctly
[image]

hankecnc, 2021-11-12:
Test code
%
( Machine Name: CNC )
( Tu Hao: _model1.prt )
( BianCheng:Administrator  )
( Time:Thu Nov  4 17:25:51 2021 )
G40 G17 G49 G21 G94 G54 G80

(cheng xu duan: :DRILLING )
(Tool Name:MILL Tool Type:Milling Tool-5 Parameters)
(Tool parameter:D=20.00; L=50.00; R=0.00; RS=2)
G90 G54 G0 X-23.44 Y-28.627 M03 S660
G43 Z10. H01 M08
G98 G82 X-23.44 Y-28.627 Z-47.5 R3. F2506. Q2.969
X-52.644 Y-14.218
X-64.364 Y20.75
X0. Y0.
X-40.732 Y24.401
X-17.292 Y37.85
X-43.229 Y51.683
X-75.315 Y40.732
X-94.912 Y-4.995
X-29.396 Y-5.956
G80
M05
M09
G91 G28 Z0.
G91 G28 Y0.
M30
(total Time=0.24)
%

hankecnc, 2021-11-12:
Should get drawn correctly
[image]

Hello, I am using GRBLHAL written by terjeio, based on F446RET6. Since the G code transmitter written by terjeio is based on C#, it cannot be used on the Raspberry Pi. I have been using your Candle2, which is very good.

Schildkroet, 2021-11-12:
Hello,
i will add this to my feature list

hankecnc, 2021-11-15:
> 您好， 我会将其添加到我的功能列表中

Thank you very
[image]
 much, it is great, G81, G82, G83 are drawn correctly,

hankecnc, 2021-11-15:
Is the poor G73 forgotten by you? You didn't even draw it?
[image]

%
( Machine Name: CNC )
( Tu Hao: _model1.prt )
( BianCheng:Administrator  )
( Time:Mon Nov 15 09:40:27 2021 )
G40 G17 G49 G21 G94 G54 G80

(cheng xu duan: :DRILLING )
(Tool Name:MILL Tool Type:Milling Tool-5 Parameters)
(Tool parameter:D=11.00; L=50.00; R=0.00; RS=2)
G90 G54 G0 X70.706 Y77.908 M03 S11110
G43 Z110. H01 M08
G98 G73 X70.706 Y77.908 Z52.5 R103. F2510. Q1.979
G80
M05
M09
G91 G28 Z0.
G91 G28 Y0.
M30
(total Time=0.03)
%

closed 2021-11-15 by hankecnc

Schildkroet, 2021-11-15:
G73 is not supported right now