Schildkroet/Candle2 #23: Interface Elements Too large for 1366x768

cefn, 2021-11-16:
Hi all,

The laptop I'm using to drive my CNC3018 is a 1366x768. 

Unfortunately some layout or scaling decisions in Candle2 mean that I can't actually fit the window in the screen in some configurations. 

In particular if I try to edit a heightmap, there is a huge panel element above (full of whitespace) which pushes the Heightmap panel out of sight, so I don't know what controls it has, as well as extending the vertical window size so I can't scroll to the bottom on the right either. 

It's possible that I can see all the controls, but I can't see the bottom of the panel so it's hard to be confident.

Is there any way to configure Candle so it is more conservative with space, or so it has scroll panels when elements are too big?

Setting the font size to 7pt doesn't proportionally scale the other elements although the text elements get smaller, so there must be some other parts which have a scale beyond my control.

Am I the only one who uses an older laptop to drive a CNC? Did I miss some configuration option which would ensure that the interface will fit.

[image]

cefn, 2021-11-16:
I was able to work around the issue by using Gnome Tweaks and radically reducing the size of system fonts.

bullestock, 2023-11-12:
@cefn you might want to check out the latest version now that https://github.com/Schildkroet/Candle2/pull/49 has been merged.