Schildkroet/Candle2 #25: Build instructions for windows 10

hunabapr, 2021-11-25:
I've installed QT 5.15.2 MinGW, 32 bit and 64 bit version.  From which I've opened Candel2 and tried to build a release version.
For both MinGW 32 & 64 bits I get 46 issues reported.

Now, as there are no issues reported to do with build / installation here it would suggest to me my problems have something with my configuration.   Any ideas?  Is there a video telling me the steps for installation etc.....

I've never used QT. I'm using a relativly clean laptop with windows 10 installed, I also have Visual studio and Arduino IDE istalled.

Thanks in advance for your response.

hunabapr, 2021-11-25:
I had athought and added the path to MinGW in my system settings wgich removed all issues execpt for this below

LICENSE<a href="https://github.com/Schildkroet/Candle2/"><span style=" text-decoration: underline; color:#0000ff;"><br/></span></a>© 2015-2016 Hayrullin Denis Ravilevich<br/>© 2018-2021 Patrick F.</p><p>Visit: <a href="https://github.com/Schildkroet/Candle2/"><span style=" text-decoration: underline; color:#0000ff;">https://github.com/Schildkroet/Candle2/</span></a></p></body></html>, 2.2.0
QMetaObject::connectSlotsByName: No matching signal for on_cmdRefresh_clicked()
16:06:35: The program has unexpectedly finished.
16:06:35: The process was ended forcefully.
16:06:36: C:\Users\The_M\Downloads\Candle2-master\build-candle2-Desktop_Qt_5_15_2_MinGW_64_bit-Debug\debug\Candle2.exe crashed.

ondabeach, 2021-11-25:
I just installed QT as well, but I installed the latest version, is that Ok or do I need to install QT 5.15.2?

Schildkroet, 2021-11-26:
Hello,
do you only get "warnings" (no errors)?
This is "normal" behaviour. The original project was started many years ago and is using code that is now considered deprecated, hence you get some warnings.

hunabapr, 2021-11-26:
Hi,
I get 45 issues, below are just two examples.  I do get an output but it doesn't give a working application (and I'm running it through the windeplyqt process) whereas if I use the last version deployed this works OK for me.  I wanted to tweak the jogging and have it read linear encoders hence why I'm trying to load the original code.

C:\Users\The_M\Downloads\Candle2-master\src\parser\gcodeparser.cpp:427: warning: variable 'j' set but not used [-Wunused-but-set-variable]
..\src\parser\gcodeparser.cpp: In member function 'PointSegment* GcodeParser::handleGCode(float, const QStringList&)':
..\src\parser\gcodeparser.cpp:427:16: warning: variable 'j' set but not used [-Wunused-but-set-variable]
         double j = qQNaN();
                ^

C:\Users\The_M\Downloads\Candle2-master\src\frmmain.cpp:1447: warning: 'int QTime::elapsed() const' is deprecated: Use QElapsedTimer instead [-Wdeprecated-declarations]
..\src\frmmain.cpp:1447:52: warning: 'int QTime::elapsed() const' is deprecated: Use QElapsedTimer instead [-Wdeprecated-declarations]
         qDebug() << "Drawn false:" << time.elapsed();
                                                    ^

Schildkroet, 2021-11-26:
You can ignore these warnings. The dont cause any problem.
On win10 i use QT 5.12.2. Maybe there is a problem with newer versions.

hankecnc, 2021-11-27:
> 您可以忽略这些警告。不要造成任何问题。
> 在 win10 上，我使用 QT 5.12.2。也许新版本有问题。

There is no problem with win10, qt5.99

hunabapr, 2021-11-27:
HI,
I've dropped back to QT 5.12.2 and it complies OK with the 32 bit version, not tried 64 bit yet.

Many thanks for your help.   You may want to amend the instructions for the windows build from 5.15.2 to 5.12.2

closed 2021-11-27 by hunabapr