Schildkroet/Candle2 #26: No machine state/position feedback to MPG

ondabeach, 2021-11-27:
Hi, I have my MPG sending commands to Candle2 and they are successfully passed onto Grbl, so that's great. But, there is no machine state/position data being sent back to the MPG. My MPG has a screen so I want to be able to see the Axes' positions and machine state.

Basically if Candle2 can just echo the information it receives from Grbl and send it to the MPG port that would give me everything I need.

IanHun01, 2022-02-15:
Try adding this at line 38 of processresponce file. I found it worked for me even when in run mode.

if(m_serialHandWheel.isOpen()) {
// write feedback
QString str(data);
QByteArray br = str.toUtf8();
m_serialHandWheel.write(br);
}

Schildkroet, 2022-07-17:
This feature is currently not planned, but you can for example echo the received data in https://github.com/Schildkroet/Candle2/blob/master/src/frmmain.cpp#L695
to the serial interface **m_serialHandWheel**. Then you can process the data as you wish

closed 2022-07-17 by Schildkroet