Schildkroet/Candle2 #30: graphical issues wayland

unrealBob, 2022-03-11:
Hey, 

I have noticed some differences regarding wayland and xorg sessions. 
When using wayland, the tool path + coordinate system, etc. doesnt show up.
Is there a way to fix this?

Wayland:
[image]
Xorg:
[image]

System:
Candle 2 version 2.4 
ThinkPad E14 gen 2 (AMD Ryzen 5 4500U, Radeon graphics) 
Arch Linux 5.16.13
KDE Plasma 5.24.3

Schildkroet, 2022-07-17:
There was a config file missing. I updated the download. 2.4 should run now.
Can you provide more infos? On Ubuntu 22.04 it runs on Wayland.

777Raim, 2022-07-18:
Yes, It runs now. Thanks!

But I can't understand, why self build Candle has wayland bug?

ps. some ` QT_DEBUG_PLUGINS=1` output info in attach (Candle 2.4)
[pre_build.log](https://github.com/Schildkroet/Candle2/files/9129897/pre_build.log)
[self_build.log](https://github.com/Schildkroet/Candle2/files/9129898/self_build.log)

Schildkroet, 2022-07-18:
I build with oldest supported Ubuntu (currently 18.04). Maybe this is a problem with newer QT version?
I will try building with a newer system and check.

777Raim, 2022-07-18:
Found some info: if run Candle with `--platform xcb` or `export QT_QPA_PLATFORM=xcb`, the bug is disappear.

Installed Ubuntu 18.04 on VBox and build Candle - on Ubuntu Wayland Candle works perfect. But on Fedora this Candle (builded on Ubuntu) has same wayland bug

Schildkroet, 2022-07-21:
Can you test this build, if it runs for you:

https://github.com/Schildkroet/Candle2/actions/runs/2713737955

777Raim, 2022-07-22:
Tried this build - AppImage runs perfect without bug

closed 2022-07-22 by Schildkroet