Schildkroet/Candle2 #34: esp32 grbl version of ethernet control can't connect properly

hanke-cnc, 2022-06-25:
Sorry, I'm not very good at programming, I can use ioSender to control the Ethernet connection normally, but candle2 never works, I tried to modify the port number of the issue here https://github.com/Schildkroet/Candle2/issues/8 ( grblesp32 default port number 23), and modifying the default configuration file port number of grbl esp32 to 30501, have always failed to succeed.

[image]
[image]

hanke-cnc, 2022-06-25:
(Google translation may be a bit problematic), I have tried 2 ways, one way is: modify the port number of the problem here https://github.com/Schildkroet/Candle2/issues/8 (grblesp32 default port number 23) , another way is: modify the default configuration file port number of grbl esp32 to 30501, but it still fails to succeed.

hanke-cnc, 2022-06-25:
Hope professional programmers can help me, (I am learning qt network programming in order to solve this problem)

Schildkroet, 2022-06-25:
Hello,
can you pls open a command window and try to ping the device (with correct IP)?
e.g. `ping -n 1 -w 1500 192.168.1.102`
If this command fails, your network configuration is invalid and must be fixed first.
Second, GRBL-Advanced creates its own TCP-Server (through Wiznet IC). Without it, Candle2 (Client) can not connect

hanke-cnc, 2022-06-26:
> Hello, can you pls open a command window and try to ping the device (with correct IP)? e.g. `ping -n 1 -w 1500 192.168.1.102` If this command fails, your network configuration is invalid and must be [...]

Hello, I use a "network debugging assistant software" and it can also connect to grbl esp32 normally. I need your help. Maybe you only need to change the source code of candle2 to achieve success
[image]
.

hanke-cnc, 2022-06-26:
Looking forward to your reply,

Schildkroet, 2022-07-22:
#35

closed 2022-07-22 by Schildkroet