Schildkroet/Candle2 #38: Height Map 1 vs. 2

Guy777D, 2022-09-12:
The only way I got heightmap to work is generate using Candle and then use the file in Candle2, so the bug is in the generation side.

Witawat, 2022-10-21:
me too error gcode some thing error not run

hobbymachinist, 2023-05-21:
Tried Candle2 since I've been using Candle for a while, and I have the same issue. I get an Error: 22 on G38.2Z-3 and subsequently the tool just moves through the grid without probing. Hope that helps debug the issue.

```
Received offsets: "[G54:0.000,0.000,0.000]; [G55:0.000,0.000,0.000]; [G56:0.000,0.000,0.000]; [G57:0.000,0.000,0.000]; [G58:0.000,0.000,0.000]; [G59:0.000,0.000,0.000]; [G28:-183.000,-195.000,-15.838]; [G30:-183.400,-195.500,-16.477]; [G92:-183.000,-172.000,-37.073]; [TLO:0.000]; [PRB:-183.000,-172.000,-37.063:1]; ok"
Updating heightmap grid drawer
Updating interpolation
Generating probe program
Updating parser: GCodeTableModel(0x7ffe087e2590) GcodeDrawer(0x5649deb9c000)
reseting gp QVector3D(nan, nan, nan)
Update parser time:  4
preparing vectors GcodeDrawer(0x5649deb9c000)
lines count 79
Updating heightmap grid drawer
Updating interpolation
Generating probe program
Updating parser: GCodeTableModel(0x7ffe087e2590) GcodeDrawer(0x5649deb9c000)
reseting gp QVector3D(nan, nan, nan)
Update parser time:  6
preparing vectors GcodeDrawer(0x5649deb9c000)
lines count 79
tool missed: 0 0 0
Error:  22
Updating interpolation
Error:  22
Updating interpolation
Error:  22
Updating interpolation
Error:  22
tool missed: 0 11 11
Updating interpolation
```

hobbymachinist, 2023-05-21:
@Schildkroet would you be able to review #47  thanks ?

PR #47 (2023-06-03): Update frmmain_heightmap.cpp to get G38.2 working
It looks like GRBL 1.1 is complaining about missing feedrate with G38.2

It seems to work with the added feedrate but it doesn't display a heat map (I'm not familiar with the code, but can take a look)

[image]

Fixes #38 partially
closed 2023-06-03 by Schildkroet