Schildkroet/Candle2 #40: Work coordinates change

rchyla, 2022-12-13:
I am using atmega8 arduino based controller with GRBL1.1 i noticed after few broken tips, why they're breaking. When you interrupt milling using STOP button, working coordinate for Z is changing to something else after 2 or 3 seconds.

gkatev, 2022-12-24:
Hi, I've noticed in my machine that when I stop on the program, the working coordinates (all of X,Y,Z) get set to the machine coordincates (which are not zero-ed properly). It this also what's happening in your scenario? (disclaimer: I'm new to candle/grbl)

rchyla, 2022-12-24:
Hello,

I currently don't have this makeshift CNC at hand, but usually the Z coordinate was changing to something around -27 after STOP (depends how high the head was), while X and Y usually stays the same according to where CNC stopped. I think that i was breaking tips because machine wasn't properly zeroed, but when it was, the head was trying to go higher than it is possible when clicking SAFE button after STOP. I didn't notice this behavior in other GRBLing software (UGS ithink).

No, the working coordinates wasn't changing to machine coordinates after stop.

Are working coordinates sent to the GRBL controller as destination and processed inside or is GRBL controller receiving machine coordinates processed by the software?

Terak8510, 2023-02-05:
Been there, and stop button does not work out so good in middle of move command.
have to hit reset and pray it stops in time.
I have to hit origin button then safe after interupting or finishing the cnc job.