Schildkroet/Candle2 #44: Customization of shortcuts and saving of initial settings

Cassotto, 2023-04-06:
Is it possible to have custom shortcuts that are saved on startup?
Thank you very much for the excellent work done.

Schildkroet, 2023-05-21:
You can use User commands. Or do you mean something different?
[image]
[image]

FabLabBaramati, 2025-05-13:
I would like to kindly request the reintroduction of the **User Command (Macro) Buttons** feature that existed in the earlier version of Candle.

These buttons were very helpful for simplifying repetitive workflows, especially in PCB milling and laser operations. In my case, I’m working with a laser head mounted on the Z-axis, and having quick access to G-code macros helps significantly with tasks like:

- Focusing the laser
- Setting work offsets
- Jogging specific coordinates
- Running test pulses
- Moving between predefined positions

To fully support my workflow, a **minimum of 17 user-configurable buttons** would be ideal. These buttons would ideally:
- Be visible and accessible in the main UI
- Allow custom G-code assignment via a settings file or GUI
- Optionally support placeholders for live coordinates (e.g., `{X}`, `{Z + 5}`)

Would it be possible to consider adding this to Candle2?  
[image]
[image]

[image]