Schildkroet/Candle2 #52: Candle2

FishmannJ, 2023-10-10:
Problémy se spouštěním ve windows 11. Po spuštění jako správce systému se zobrazují přiložená hlášení. [](url)
[image]
[image]

bullestock, 2023-11-12:
@FishmannJ in English, please?

FishmannJ, 2023-11-13:
Startup problems in windows 11. After running as system administrator, the attached messages are displayed. Picture attached in the previous message.

Schildkroet, 2023-11-14:
Hello,
there is currently no setup available for latest windows (since i only use linux).
I will try to setup a new workflow for latest windows in the near future.

FishmannJ, 2023-11-14:
Mnohokrát děkuji. Klement

> Od: "Schildkroet" ***@***.***> 
> Komu: "Schild.2"C4hubre.andle> 
Komu: "Schildkroet"2com. 11.2023 10:40 
 
Dobrý den, 
v současné době není k dispozici žádné nastavení pro nejnovější okna (protože používám pouze linux). 
V blízké budoucnosti se pokusím nastavit nový pracovní postup pro nejnovější okna.
— 
Odpovězte přímo na tento e-mail, zobrazte jej na GitHubu <https://github.com/Schildkroet/Candle2/issues/52#issuecomment-1809861714> nebo se odhlaste z odběru <https://github.com/
Dostáváte to, protože jste byli zmíněni.ID zprávy: <Schildkroet/Candle2/issues/52/1809861714 @ github . com>

Schildkroet, 2024-03-06:
Can you please test the build:

https://github.com/Schildkroet/Candle2/actions/runs/8172000331

commit 26249f0d (2024-03-06): Update build_win.yml. Fixes #52
closed 2024-03-06 by Schildkroet