Schildkroet/Candle2 #62: Visualizer problem

ddllpp, 2024-09-16:
Using MX Linux 23.3 on an HP Elitebook 2760p.  The Candle2 AppRun opens up the GUI, but I see the desktop in the visualizer window.  
Checking OpenGL version with 
glxinfo | grep "OpenGL version" 
response is:
OpenGL version string: 3.3 (Compatibility Profile) Mesa 22.3.6

Trying the mesa override..
 sudo MESA_OVERRIDE=3.3 ./AppRun 
also shows the desktop in the visualizer.

Can anyone help?

Here's the text in the terminal window:
`locale: "en"
style "Windows"
style "Fusion"
QString::arg: Argument missing: <html><head/><body><p>Candle 2<br/>Version: 2.4<br/>License: GNU GENERAL PUBLIC LICENSE<a href="https://github.com/Schildkroet/Candle2/"><span style=" text-decoration: underline; color:#0000ff;"><br/></span></a>© 2015-2016 Hayrullin Denis Ravilevich<br/>© 2018-2022 Patrick F.</p><p>Visit: <a href="https://github.com/Schildkroet/Candle2/"><span style=" text-decoration: underline; color:#0000ff;">https://github.com/Schildkroet/Candle2/</span></a></p></body></html>, 2.2.0
QMetaObject::connectSlotsByName: No matching signal for on_cmdRefresh_clicked()
shader program created
preparing vectors GcodeDrawer(0x55e0788d0bb0)
lines count 0
preparing vectors GcodeDrawer(0x55e0788d0f40)
lines count 0
`

ddllpp, 2024-09-17:
Solved this with the following prefix to the AppRun:
'LIBGL_ALWAYS_SOFTWARE=1 ./AppRun'
The visualizer appears all black, but there is a UCS icon and once you change the colors in settings, everything is fine.

alligitor, 2025-05-19:
I just compiled Candle2 on a Lenovo ThinkCenter and ran into this same problem.  In my case, the visualizer area show the content of some window behind Candle2.

setting 
LIBGL_ALWAYS_SOFTWARE to 1 or true 
did not make a difference.

System Spec
CPU: Intel Core i5-7600T
Graphics: VGA compatible controller: Intel Corporation HD Graphics 630 (rev 04)
RAM: 16G

this is how I ran Candle2:

Candle2/src$ LIBGL_ALWAYS_SOFTWARE=true ./Candle2

alligitor, 2025-05-20:
I tried compiling on a Raspberry Pi 4, 4G.  
Compiles, runs but visualizer area is black.

[image]

A few things.  In settings->Colors, all were black by default.  I hit "Reset to Defaults".  All colors went back to normal but visualizer is still all black.

The following is the output on console:

```
Candle2/src $ QStandardPaths: wrong permissions on runtime directory /run/user/1000, 0770 instead of 0700
locale: "en"
style "cleanlooks"
style "gtk2"
style "cde"
style "motif"
style "plastique"
style "qt5ct-style"
style "Windows"
style "Fusion"
QString::arg: Argument missing: <html><head><title>About</title></head><body><p>Candle 2<br/>Version: 2.6<br/>License: GNU GENERAL PUBLIC LICENSE<br/>Â© 2015-2016 Hayrullin Denis Ravilevich<br/>Â© 2018-2024 Patrick F.</p><p>Visit: <a href="https://github.com/Schildkroet/Candle2/"><span style=" text-decoration: underline; color:#0000ff;">https://github.com/Schildkroet/Candle2/</span></a></p></body></html>, 2.2.0
settings file path:  "/home/pi/.config/Candle2/settings.ini"
QMetaObject::connectSlotsByName: No matching signal for on_cmdRefresh_clicked()
shader program created
preparing vectors GcodeDrawer(0x555c72f2f0)
lines count 0
preparing vectors GcodeDrawer(0x55602c8860)
lines count 0
```